robotics-in-concert/rocon_concert #254: [concert_service_manager] create cache to store local configuration

jihoonl, 2015-01-12:
As follow up of https://github.com/robotics-in-concert/rocon_qt_gui/issues/170#issuecomment-69523221.

jihoonl, 2015-01-12:
From @dwlee 
Structure of cache file and temporary directory about service configuration.
```
${ROS_HOME}/rocon/<lower_case_and_underscored_concert name>/
|-<services name>.services
|-<service name>/
    |-<service name>.service
    |-<service name>.interactions 
    |-<service name>.launch
    |-<service name>.parameters
```

For example,  `Chatter concert` where ROS_HOME is `/home/dwlee/.ros`

```
/home/dwlee/.ros/rocon/chatter_concert/
|-chatter.services
|-chatter/
    |-chatter.service
    |-chatter.interactions 
    |-chatter.launch
    |-chatter.parameters
```

jihoonl, 2015-01-12:
A solution may have two different services with different configurations. If service directories are created based on resource_name it would get conflict. It should be based on instance name.

Also can you explain how `service_manager` would read cached configurations?

dwlee, 2015-01-12:
I will put the flag for deciding  cached file or not when service manager load service profiles from solution configuration.
And, I thought different structure reflected you concern.
How about as following?
```
${ROS_HOME}/rocon/<lower_case_and_underscored_concert name>/
|-<services name>.services
|-services/    
    |-<overriding service name>/
        |-<service resource name>.service
        |-<service resource name>.interactions 
        |-<service resource name>.launch
        |-<overriding service name>.parameters
```

dwlee, 2015-01-13:
* Former method
service manager used to load service configuration directly from service configuration file.
If overriding information is exigisted, service manager chaged to new infomation instead of former one at that time.
  * Privious configuration structure
    * concert_name: Chatter Concert
    * solution configuration: chatter.services
    ```
    - overrides:
        name: hi
        description: chatting with saying hi. 
        parameters: chatter_concert/hi
      resource_name: chatter_concert/chatter
    - overrides:
        name: hello
        parameters: chatter_concert/hello
      resource_name: chatter_concert/chatter
    - resource_name: concert_service_admin/admin
      ```

* Improvement method
service manager loads the service configuration information from solution and saves cache reflected overriding infromation
It launch the service with loading this cache 

  * The order is following
      * Launch a solution
      * Check whether cache existing
      * If no cache, make the cache by loading from solution base information
      * Load cache and launch a service

  * Combination each service information with former configuration as following
    ```
    ${ROS_HOME}/rocon/<lower_case_and_underscored_concert name>/
    |-<services name>.services
    |-services/    
        |-<overriding service name>/
            |-<overriding service name>.service
            |-<overriding service name>.interactions 
            |-<overriding service name>.parameters
    ```

  * Geneation cache following structure in case of above thing
    * concert_name: Chatter Concert
    * solution configuration: chatter.services
    ```
    ${ROS_HOME}/rocon/chatter_concert/
    |-chatter.services
    |-services/    
        |-hi/
            |-hi.service
            |-hi.interactions 
            |-hi.parameters
        |-hello/
            |-hello.service
            |-hello.interactions 
            |-hello.parameters
        |-admin/
            |-admin.service
            |-admin.interactions 
            |-admin.parameters
    ```
  
  * chatter.services 
    ```
    - name: chatter_concert/hi
      enabled: true
    - name: chatter_concert/hello
      enabled: false
    - name: concert_service_admin/admin
    ```
    * Addition of property about enabled
      * storing flag regarding service enabled during runtime
      * service restart using it when solution restarted

  * hi.service 
    ```
    name: hi
    description: chatting with saying hi. 
    author: someone
    priority: 0
    launcher_type: roslaunch
    launcher: <pull path of chater launcher>/chatter #absolute path
    interactions: hi # relative path
    parameters: hi # relative path
    ```
    * reason about difference of path structure among launcher, interactions and parameter
      * The information(possible to override information) changed in run time alwalys store in file to keep out of date. 
      * That information (parameter, interactions) stores to cache, they are loaded from relative path when service is launched
      * The information not changed in runtime as laucher is gotten from resource
      * They are loaded from absolut path when service is launched

  * hi.parameters
  ```
  topic_name: chatter_with_hi
  ```

  * hi.interactions
  ```
  - name: chatter_concert/remocon_dude
    role: 'Chatter'
    compatibility: rocon:/pc/*/hydro|indigo/precise|quantal|raring|saucy|trusty
    display_name: Qt Listener
    description: For human dudes who have to listen to nagging dudettes.
    max: -1
    remappings:
      - remap_from: chatter
        remap_to: /conversation/chatter
  - name: com.github.rosjava.android_remocons.listener.Listener
    role: 'Chatter'
    compatibility: rocon:/*/*/hydro/jellybean|ice_cream_sandwich|kitkat
    display_name: Android Listener
    description: Android listener 
    max: -1
    remappings:
      - remap_from: chatter
        remap_to: /conversation/chatter
  - name: web_url(http://wiki.ros.org/chatter_concert)
    compatibility: rocon:/*/*/*/chrome|firefox|safari|opera|internet_explorer
    display_name: Ros Wiki - Chatter Concert
    description: A ros chatter style introduction to the concert.
    icon:
      resource_name: rocon_bubble_icons/ros.png
  - name: chatter_concert/listener.launch
    role: 'Chatter'
    compatibility: rocon:/pc/*/hydro|indigo/precise|quantal|raring|saucy|trusty
    display_name: Roslaunch Listener
    description: Another human dude listening to dudette, this time via roslaunch.
    max: -1
    parameters:
      topic: /conversation/chatter

  ```

dwlee, 2015-01-19:
Remain issues
* [ ] Update enabled status in cache
* [ ] Update cache just changed service profile
* [ ] Create cache about default resource path list

closed 2015-01-21 by jihoonl