robotics-in-concert/rocon_concert #304: Cannot start the most basic of concerts

stonier, 2015-06-12:
This fails:

```
roslaunch concert_master concert_master.launch --screen
```

with continuous output:

```
[WARN] [WallTime: 1434090729.517977] Service Manager : resource could not be split with a valid leading package name []
[INFO] [WallTime: 1434090730.019064] Service Manager : detect changed solution config. reload
[INFO] [WallTime: 1434090730.019961] Service Manager : load service profile from default configuration
[WARN] [WallTime: 1434090730.020821] Service Manager : resource could not be split with a valid leading package name []
[INFO] [WallTime: 1434090730.522353] Service Manager : detect changed solution config. reload
[INFO] [WallTime: 1434090730.522609] Service Manager : load service profile from default configuration
[WARN] [WallTime: 1434090730.522855] Service Manager : resource could not be split with a valid leading package name []
[INFO] [WallTime: 1434090731.023683] Service Manager : detect changed solution config. reload
[INFO] [WallTime: 1434090731.023937] Service Manager : load service profile from default configuration
[WARN] [WallTime: 1434090731.024146] Service Manager : resource could not be split with a valid leading package name []
```

jihoonl, 2015-06-12:
Two options to fix this issue.

* Make an error and terminates if `services` file is not specified.
* Make dummy.services

Which one would be nicer?

stonier, 2015-06-12:
It shouldn't terminate if services is not provided...it just means none are loaded or enabled.

commit 74baecdf (2015-06-12): [concert_service_manager] bugfix concert when no services are provided, fixes #304.
PR #305 (2015-06-12): Concert should start even if there are no services by default
This fixes #304.
commit f7da454e (2015-06-12): [concert_service_manager] bugfix concert when no services are provided, fixes #304.
closed 2015-06-12 by jihoonl